floppes/RTD266xFlash #1: Connection in Flash Programm cannot be established

bluesbrother84, 2018-10-09:
Hello,

first of all thanks for this good GitHub page an explanation.
Unfortunately I stuck in a problem where I didn't find the answer by my own. I'll try to explain and maybe you have an idea.

Environment:
- I'm currently want to adjust the display to have the Zeroboy Logo on it
- The Display is already power optimized like mentioned on the Zeroboy Homepage:
https://www.zeroboy.eu/wp-content/uploads/2018/06/kedei_energiesparen_mod.jpg
- A bridge is created between pin1 and pin2 on the power switch
https://www.zeroboy.eu/wp-content/uploads/2017/03/diagram_power.jpg
- The display itself works fine and can be used. Screen of RetroPi is shown up
- The Arduino Sketch was moved to the Arduino Uno and connected at the corresponding Pins to the HDMI Port.
- The Screen is switched on first, then the Arduino is connected, flashlight is active and then the Flash Program is started

Problem:
- I select the correct COM Port and press connect. The only response I get is 'error'
- When I start to debug the code with Visual Studio I've found the point where it stopped (in file RTD266x.cs -> Method public Result ReadErrorCode
The ReadComPort Method is successfully called, the Com Port is not null and open. Looks fine but the _comPort.ReadByte(); will returns results which are not valid

Result result = ReadComPort(7, out response);
if (result != Result.Ok) -> This statement is okay
...
if (response[0] != (byte)RtdCommand.GetError) -> This statement end in error
{
   return Result.UnexpectedCommand;
}

The array looks like this
[0] = 111	should be 7 
[1] = 118	should be 0
[2] = 101
[3] = 114
[4] = 102
[5] = 108
[6] = 111	

When I try it again the numbers will change but never be as expected. What was also strange even if no screen is connected I'm getting a result.

As I'm not that experienced in C# and the Com Ports my knowledge ends. First Google results also didn't help me to solve this. Further I've also no brand new display to check if this is related to the adjustments I've done. Maybe you have an idea how to solve this. As I've also checked in the ZeroBoy Facebook group it seems I'm the only one with this issue as of now.

Already tried:
- I've added comments in the Arduino Sketch after the major points in the code and watched them in the Serial Monitor. From this perspective it looks like the Arduino itself does not have a problem
- I've tried different start/connection methods between Laptop, Arduino and Display

Thanks already in advance for your help :)
Best regards
Chris

floppes, 2018-10-10:
A flashing LED on the Arduino indicates a connection problem. The GUI tool tries to read the error code in `btnConnect_Click` but this already fails in your case. You should check the connection again. Sometimes the HDMI connector on the display is very sensitive. On one of my boards I had to use a lot of power to plug my HDMI plug into the connector to make it work.

The data you see in `response` reads **overflo** when you convert the bytes to ASCII characters. Did you add any output code to the Arduino sketch (`Serial.write` or `Serial.print`)? The original code does not print **overflo** to the serial connection. Any additional output will break the communication between Arduino and the GUI tool

floppes, 2018-10-23:
There either seems to be a problem with the virtual COM port (but it worked with your debug outputs?) or the Arduino is stuck at some place.

I suggest you put some debug output into the Arduino function `setup` and use the Arduino IDE's serial monitor to display them. Check if it reaches the end of `setup` and if `loop` is executed.

timkn, 2018-11-05:
I seem to have the same problem, LED not blinking, Serial is working with debug-code in setup() and loop().
I2C Scanner is showing three device IDs. When trying to connect with your programmer I just get "error".

bluesbrother84, 2018-11-05:
Hi,

thanks for the hint. Tried it right now with baud rate 9600 and it connects. Flashing works as well.
Problem solved. Perfekt :+1: 

Thanks a lot for your help.
Best regards
Chris

closed 2018-11-05 by bluesbrother84

timkn, 2018-11-06:
I have changed them both. When using 115200 baud it didn't work, haven't checked any other baudrate.

bluesbrother84, 2018-11-06:
I did the same in GUI and Arduino.

VeteranGamer, 2018-11-07:
i'm also facing the same issue....
tried 115200 (flashed Arduino) and selected in GUI.... no success
tried 9600 (flashed Arduino) and selected in GUI.... no success

also tried to change the Bit Rate for the port.... no success

disappointing....

.

krisstakos, 2021-02-28:
I'm facing the same issue. Uploaded arduino i2c and it's finding a lot of devices:

> Scanning...
> I2C device found at address 0x37  !
> I2C device found at address 0x3A  !
> I2C device found at address 0x4A  !
> I2C device found at address 0x4B  !
> I2C device found at address 0x50  !

In the gui is retrying 5 times and cannot connect. Tried 115200 also 9600 same effect. Any help? After many months ago I successfully connected them but not know.. I'm using arduino pro micro