floppes/RTD266xFlash #2: Unknown firmware

TRIGGERhappyTV, 2019-02-13:
Hello,

First of all many thanks for the good Github page.
Everything is explained very well.

I have only a small problem.

I have downloaded and run the RTD266xFlash.exe v1.5.0.0. If I Select "Remove" HDMI "pop-up" and click on "Modify Firmware" everything goes through to 100% and then I get the error message "checking firmware ... You can send your firmware to the author, maybe it can be added to the known firmwares. "

I'm really sorry, but unfortunately I'm almost a total beginner. Both with the Arduino and Github. Unfortunately I can not use more than your program. For that I lack the knowledge.

With the attached Dropbox link you get to backup my firmware for the KeDei 3.5 inch display.

https://www.dropbox.com/s/236z1o7zqnhgmch/firmware-2019-02-13-20-39-53.bin?dl=0

Please excuse the circumstances.
I hope you can help me.

Thank you in advance.

Peter

floppes, 2019-02-14:
Hi Peter,

thank you for the firmware. I haven't seen this variant before but it looks very similar to the others. I have added it to the list of known firmwares. You can try a preview of version 1.6.0.0 here:

[RTD266xFlash_v1.6.zip](https://github.com/floppes/RTD266xFlash/files/2866679/RTD266xFlash_v1.6.zip)

Please tell me if this works with your firmware.

It would be very helpful if you could upload a picture of the back side of your panel so I can see the cable and the print on it. This helps me to identify the type of panel.

TRIGGERhappyTV, 2019-02-16:
Hi floppes,

many thanks for the preview of the new version and the fast implementation !! : D

Works perfect.
You're great (thumbs up).

I have attached pictures afterwards.

Unfortunately, I have already installed and wired the screen. Hopefully you can see what you need.

Thanks again for your effort.

[image]

[image]

floppes, 2019-02-17:
That's great to hear!

Do you have a picture of the backside of the panel? I'd like to see the printing on the cable, like this:

[image]

TRIGGERhappyTV, 2019-02-17:
Unfortunately not. Sorry. 

The board is glued to the panel with Powerstips. I can not raise the board high enough for me to see the inscription without possibly damaging the panel :/

closed 2019-02-17 by floppes