floppes/RTD266xFlash #4: JEDEC ID: 0x856013 Cannot setup chip commands. The flash chip id is unknown.

korbav, 2020-12-01:
Hello, first of all, thanks for the awesome job, the following is not quite an issue but much more an information request.

I'm trying to get my 7 inches Kedei flashed (using the RPi method), this screen is also managed by a RTD2660 chip.
When I launch the python script it stops with the following error message : 

```
JEDEC ID: 0x856013
Cannot setup chip commands. The flash chip id is unknown.
```

The script might not be ready for this device but I'm wondering if by any way I could make it work with my device?

Below are some pictures of the board in the case it brings more information to your knowledge : 

[image]
[image]
[image]

floppes, 2020-12-01:
Hi korbav,

I can't really identify the EEPROM chip on your photos. It should be the one below the small flat flex cable.
They come from different manufacturers, yours looks like STMicroelectronics. They all usually speak the same protocol. You can simply try additing the manufacturer ID 0x85 in `RTD266xPy/rtd266x/rtd266x.py` line 204 and see if it works.

korbav, 2020-12-02:
Hi floppes, thanks for your answer.

I just gave a try with your instructions and it worked well, the firmware can be extracted.

I then tried to modify it using the exe and faced this error message : 

```
Error! Unknown firmware. You can send your firmware and the name of the display (printed on the cable on the back side) to the author (floppes@gmx.de), maybe it can be added to the known firmwares.
```

Here is the firmware that has been output : [firmware](https://bit.ly/3qlyOgs)

It would be awesome if you could take a look or if you could give me the instructions to add this firmware.
The model of the device is KD070V02, the brand name is KeDei, based on a RTD2660 chip.

floppes, 2020-12-02:
So far I have only had 3.5" displays from KeDei with a resolution of 480x320. They came with lots of different panels and firmwares and I analyzed only those firmwares for my hardware.

I don't have your board, but your firmware looks very similar to the ones I know. You can add the following code to RTD266xFlash\RTD266xFlash\Firmware.cs to make it known to the tool:

```
new Firmware("7\" KeDei KD070V02 1024x600", 0x260D8, 1507, 0x12346, 0xD04F, 0x14733, 0x13A31, 0x15577, new[]
{
	new HashInfo(0, 0x80000, "6EEF496354199EA4147F6EF5589194D7FD547CC43BC33E3567A0D2411A84C5AC", new []
	{
		new HashSkip(0xD04F + 0x1D, 48),
		new HashSkip(0x12346, 16),
		new HashSkip(0x13A31, 48),
		new HashSkip(0x14733, 1),
		new HashSkip(0x15577, 1),
		new HashSkip(0x260D8, 1507)
	})
})
```

You can try to modify your firmware with this. But especially for the boot logo, it may not work properly because of the different screen resolution.

With the RTD266x there is no real risk to brick your display. Even if you flash an invalid firmware, you can always reflash another. Just make sure you have a copy of your original firmware ;)

korbav, 2020-12-02:
Thanks for your assistance, it did work well! Even the logo worked well, probably a bit small but you had foreseen it would happen since the project stands initially for lower resolutions.
It might be interesting for other people to update the project to handle this firmware, if you wish I can create a PR.

floppes, 2020-12-03:
That's great, thanks for the feedback 👍
I have added the firmware and EEPROM manufacturer ID and published a new version. No need for a PR anymore ;)

closed 2020-12-03 by floppes

korbav, 2020-12-05:
> 
> 
> That's great, thanks for the feedback 👍
> I have added the firmware and EEPROM manufacturer ID and published a new version. No need for a PR anymore ;)

Awesome, thanks!

Shackmeister, 2021-01-04:
Hi @floppes 

With newest version (2.6.0.0) I'm also getting the same error in the gui "Initialization error: could not detect chip type, unknown JEDEC id 0x856013". Am I doing something wrong or is this perhaps a bug?

Shackmeister, 2021-01-04:
Datasheet and image of the EEprom is attached below:
https://datasheet.lcsc.com/szlcsc/1810010235_PUYA-P25Q40H-SSH-IT_C194873.pdf
[image]

floppes, 2021-01-06:
Right, I forgot to add the full JEDEC id to the Arduino project. korbav used the Python script without the Arduino, that's why it worked for him.
Can you please update your Arduino sketch and try again?

Shackmeister, 2021-01-06:
Thanks Floppes, I'll get it tested right away.

SparkyNZ, 2021-09-04:
Does anyone know where I can find a schematic/gerber file for this KD070V02 board?? I have sent an email to KeDei but I don't like my chances of getting a response.

I tried modifying the HDMI connector and the traces have come off the board.. so what I want to do is redesign the layout of the board so it will fit inside my nav unit casing in my car.