floppes/RTD266xFlash #7: I get the error "uknown firmware" when trying to load the software through python

hell-boy420, 2021-02-20:
help im not really sure what to do. Did i just happen to get a non-compatible board? </3

floppes, 2021-02-20:
More info from your side would be helpful.

- Name of your board
- Picture of your board
- Upload of your firmware

hell-boy420, 2021-02-20:
[image]
Thank you so much for the response! The model is: KD0350AV02. Its listed on Amazon under the brand Osoyoo, "Osoyoo 3.5" HDMI LCD Touch..." By upload of firmware do you mean the out.bin file? Thank you for the help!

floppes, 2021-02-20:
That's a new board variant I haven't seen before. What is the physical resolution of the panel?
Yes, please upload out.bin.

hell-boy420, 2021-02-20:
Sorry for the late response, it was late and i went to bed. Amazon says the "physical resolution" is 800x480 and the "Original resolution for the LCD is 480x320." I'm not to adept with GitHub and I'm unsure how to attach the out.bin file. Should I upload it and then create a link, or is there an easier way?

floppes, 2021-02-20:
You can zip it and attach it by dragging & dropping it onto the input form.
You could also upload it somewhere and post the link here.

hell-boy420, 2021-02-20:
[out.zip](https://github.com/floppes/RTD266xFlash/files/6016126/out.zip)
Okay, again thank you for the help!

floppes, 2021-02-20:
Thanks, the firmware looks very similar to the ones I already have. I've added it to the tool and released v2.7.0.0. You can give it a try and see if it works for you.
Always keep the original firmware in case something goes wrong ;)

hell-boy420, 2021-02-20:
Thank you so much! It works flawlessly :)

floppes, 2021-02-20:
Great :)

closed 2021-02-20 by floppes

Gand46, 2021-09-11:
Hi i have the same board but it gives me the same problem with the GUI tool =  Checking "firmware... Error! Unknown firmware. You can send your firmware and the name of the display 

[KD0350AV02.zip](https://github.com/floppes/RTD266xFlash/files/7147045/KD0350AV02.zip)