floppes/RTD266xFlash #8: Cannot connect to gui tool

krisstakos, 2021-02-28:
I'm facing the same issue a few folks had. Uploaded arduino i2c scan and it's finding a lot of devices:

    Scanning...
    I2C device found at address 0x37 !
    I2C device found at address 0x3A !
    I2C device found at address 0x4A !
    I2C device found at address 0x4B !
    I2C device found at address 0x50 !

In the gui it's retrying 5 times and cannot connect. Tried 115200 also 9600 same effect. Any help? After many months ago I successfully connected them but not now.. I'm using arduino pro micro

floppes, 2021-02-28:
What's the status of the Arduino's LED?
You need to power the display first, then the Arduino.
Do you have an Arduino Uno and can you try with it?
Is the Arduino's LED flashing when you try to connect with the GUI tool?
Which operating system are you using?
Did you select the correct COM port?

krisstakos, 2021-02-28:
It flashes once every retry. The display is powered on prevously. From he i2c scan I can see bunch ob devices from the display board. No, I don't have arduino UNO to try. I'm using win10 64-bit. Yes, I do selected the correct COM port

krisstakos, 2021-02-28:
Succeeded with RPI but not with arduino way too many issues

floppes, 2021-02-28:
Ok I can reproduce the problem with a SparkFun Pro Micro. The GUI tool won't connect.
But luckily I found a solution. You need to add

```
_comPort.DtrEnable = true;
_comPort.RtsEnable = true;
```

in FormArduino.cs, function btnConnect_Click() after `_comPort.Handshake = Handshake.None;`

Or apply this patch:

```
diff --git a/RTD266xFlash/RTD266xFlash/Forms/FormArduino.cs b/RTD266xFlash/RTD266xFlash/Forms/FormArduino.cs
index 4cff805..3d0208b 100644
--- a/RTD266xFlash/RTD266xFlash/Forms/FormArduino.cs
+++ b/RTD266xFlash/RTD266xFlash/Forms/FormArduino.cs
@@ -203,6 +203,8 @@ namespace RTD266xFlash.Forms
             _comPort.StopBits = StopBits.One;
             _comPort.DataBits = 8;
             _comPort.Handshake = Handshake.None;
+            _comPort.DtrEnable = true;
+            _comPort.RtsEnable = true;
 
             AppendConsoleText("Connecting...\r\n");
 
```

Can you please try this and report back?

krisstakos, 2021-03-03:
I guess I should compile the gui again?

krisstakos, 2021-03-03:
Recompiled with the patch applied and now is working. Thanks.

floppes, 2021-03-05:
Great :) I've released an updated version with the fix.

closed 2021-03-05 by floppes