floppes/RTD266xFlash #9: self.i2c = SMBus(interface) IOError: [Errno 2] No such file or directory

Electronlibre2012, 2021-04-29:
Hello,

sorry to disturb you but i try your tutorial with a RPI4 and i have this error : 
` python rtd266x_flash.py -r 524288 out.bin
Traceback (most recent call last):
  File "rtd266x_flash.py", line 41, in <module>
    rtd = RTD266x(args.interface, 0x4A)
  File "/home/pi/RTD266xFlash/RTD266xPy/rtd266x/rtd266x.py", line 24, in __init__
    self.i2c = SMBus(interface)
IOError: [Errno 2] No such file or directory
`

the first issue is i cant detect anything with `sudo i2cdetect -y 2` only `sudo i2cdetect -r -y 11
` and then i found the 4A I2s module.

so i found with googling : in /boot/config.txt look for the [pi4] section and set this value dtoverlay=vc4-kms-v3d,
delete the "f" of fkms, and then i was able to `sudo i2cdetect -r -y 11` and found the 4A I2s of the screen but the above error is when i try to read the flash memory with `python rtd266x_flash.py -r 524288 out.bin`.

Any idea please?

Thanks!

floppes, 2021-04-29:
A quick Google search for `smbus No such file or directory pi 4` returns some useful hints like running `sudo raspi-config` and enabling I2C. Did you try this already?

Electronlibre2012, 2021-04-29:
Thanks for your fast reply!

look at the EDIT of my post...find the docs ;)

thanks! and nice work!

floppes, 2021-04-30:
Thanks for the solution!

closed 2021-04-30 by floppes