floppes/RTD266xFlash #10: Support for Waveshare 4.3"

thomasgoedertier, 2021-08-23:
Hello,

First of all I want to thank you for your excellent tool and very clear instructions!
Thanks to your project I was able to disable the 'HDMI popup' on a [Waveshare 4.3" display](https://www.waveshare.com/4.3inch-hdmi-lcd-b.htm). 
This display also uses the RTD2660 chip.
I have attached the original.bin read from this display.

I found the 'HDMI popup' function at 0x13FE00. By changing the value from 0xE4 to 0x22 I was able to disable the popup.

I also found the 'HDMI popup text' (0x19 0x14 0x1E 0x1F 0x1A) at 0x1233F. However I did not try to change this as I did not need it. The default text shown in the HDMI popup is "HDMI-WaveShare"
Neither did I try changing the background or the logo...

Feel free to include my findings in your project.

Kind regards,
Thomas

[waveshare4_3.zip](https://github.com/floppes/RTD266xFlash/files/7029836/waveshare4_3.zip)

floppes, 2021-08-28:
Hi Thomas,

thanks a lot for your files. I also found some similarities to the 3.5" firmware variants, but not the palette and logo location. They are needed in order to be included in the GUI tool.

thomasgoedertier, 2021-09-06:
Hi floppes,

Ok no problem.
I should have mentioned that this display probably doesn't have a background color or logo function.
When it boots there is only a black screen and then the 'HDMI-popup' is shown while the raspberry pi boot screen becomes active.

You can close this issue or leave it open.. whatever suits you best.

I just wanted to document my solution.. maybe it can help someone with the same brand of display that stumbles upon your repository.

Thanks again for your work!
Kind regards

closed 2021-09-06 by floppes