floppes/RTD266xFlash #12: I get the error "uknown firmware" when trying to load the software through GUI Tool

Gand46, 2021-09-11:
Hi i have the KD0350AV02 model but it gives me a problem with the GUI tool = Checking "firmware... Error! Unknown firmware. 

i upload the firmware 
[KD0350AV02.zip](https://github.com/floppes/RTD266xFlash/files/7147049/KD0350AV02.zip)

Gand46, 2021-09-11:
the difference between mine and this [Issue #7](https://github.com/floppes/RTD266xFlash/issues/7) are the brightness control

floppes, 2021-09-11:
Thank you for the file. I have added the firmware and released a new version.

Gand46, 2021-09-11:
Thank you; it worked to change the logo png, logo color, and background logo color, but not the blue background; in any case, it looks great
[image]

floppes, 2021-09-11:
You're right, the offset for the background color was wrong. I've published a new release. Can you try again please?

Gand46, 2021-09-11:
Finally, I can make it look perfect. Thank you.

[image]

You did a great job.

closed 2021-09-11 by floppes