floppes/RTD266xFlash #13: Update readme because issues

syproduction, 2021-09-23:
I am working with PCB800099 on RTD 2660 chip and found some undocumented problems.

1. Change BAUD_RATE to 9600 in RTD266xArduino and in RTD266xFlash.exe, otherwise Connection Timeout error
2. Uncomment string:
` {"W25X40"     , 0xEF3013,      512,       256, 64},`
in rtd266x.cpp in Arduino IDE window, otherwise Unknown JEDEC error

PS. It is useful to debug connection issues using famous "i2c_detect" Arduino sketch. Should spit out something like:
```
I2C device found at address 0x37  !
I2C device found at address 0x4A  !
I2C device found at address 0x4B  !
I2C device found at address 0x50  !
```

floppes, 2021-09-30:
Thank you for your suggestions.

The baud rate should have nothing to do with the display. It just defines the speed of the serial connection between the Arduino and your computer. If 115200 baud does not work for you, there may be problems with your computer or the connection to the Arduino.

Most EEPROM types in `rtd266x.cpp` are commented to save RAM. I have updated the readme with a hint about the JEDEC ids.

Gonielc, 2021-10-09:
hi Ms floppes i have problem try to flash and this appears Initialization error: could not detect chip type, unknown JEDEC id 0xEF3013 
[image]
[image]
im new using an arduino

floppes, 2021-10-10:
You have to edit `rtd266x.cpp` and enable the line with your JEDEC id `0xEF3013`.

Gonielc, 2021-10-10:
thanks floppes