floppes/RTD266xFlash #14: Support for MPI7002 7"

almirus, 2021-10-31:
Hi!
Please add support for [new display](http://www.lcdwiki.com/7inch_HDMI_Display-H) 

If it possible please make path for this:
there is backlight adjust button, if short click the button, the brightness is added (+10).
if you hold it for more than 3 seconds, the screen turns off. to turn on the screen, press the button again.
BUT there is an error. If I want to turn off the screen, then the brightness is **added first**, then the screen turns off.
and it may happen that the next switching on of the screen will be with a brightness equal to 10

[out.bin.zip](https://github.com/floppes/RTD266xFlash/files/7448483/out.bin.zip)

floppes, 2021-10-31:
Sorry, the firmware is too different from the ones I know.
The patch is very hard without the source code or the hardware.

closed 2021-11-23 by floppes

floppes, 2022-05-27:
I dug a little deeper into the firmware structure and was able to add this firmware.
It is included in release v2.12.0.0 .