floppes/RTD266xFlash #15: Unknown firmware on latest KeDei KD0350AV02 3.5" TFT

tarbard, 2021-11-22:
Thanks for this utility, I have successfully used it to flash one of these in the past but the latest one I bought results in "Error: Unknown Firmware".    The PCB looks the same as the previous version but when I run diff on the out.bin I had dumped from the old screen compared to this it reports it is different.    I will attach the dump file and picture from my latest screen.

[out.bin.2021-11-22.zip](https://github.com/floppes/RTD266xFlash/files/7580352/out.bin.2021-11-22.zip)
[image]

floppes, 2021-11-23:
Thank you for the firmware!
It is very similar to the firmwares I already have and I will add it to the tool.
Can you read the panel type on the orange flat-flex cable so we can identify it by name?

tarbard, 2021-11-23:
Thanks for your help.   I can't see any writing on the flat cable, here is a close up picture to illustrate. 

[image]

floppes, 2021-11-23:
The printing is usually at the other end of the cable. You would have to remove the panel from the board to see it. But don't do it just for me, it's ok without the exact name ;)
I've just released v2.10.0.0 which includes your firmware.

tarbard, 2021-11-25:
Thanks so much.    I cracked my last screen so i'm handling this one with extra care :-)

closed 2021-11-25 by tarbard