floppes/RTD266xFlash #17: New JEDEC id 0x856014 won't flash

EmpressFiona, 2022-04-03:
This is related to #4 as it appears this chip (P25Q80H) is more or less identical to that one, just bigger.

I'm using Python to work with HDMI to dump the firmware from my chip, but it's not working so well. I can get it to *read* off the chip, but the data is volatile when I do - that is, I can read twice and get two slightly different files. I also am completely unable to write at all:

```
Resetting device...
Entering ISP mode...
Reading chip data...
JEDEC ID: 0x856014
ID: 0x8513
Status: 0xfc
Clearing lock bits...
Status: 0xfc
Writing data from file some_file.bin to offset 0...
Erasing sector at address 0x0
Writing 256 bytes to address 0x0
CRC validation failed! Expected 0x5, received 0xba
Error: cannot program flash!
Resetting device...
```

Is there an easy way to fix this? I checked some of the relevant code and if I didn't misread, the 0x85 at the front of the flash code *should* be enough on its own to apply the necessary fixes.

floppes, 2022-04-04:
It looks like your I2C communication is bad. There could be several reasons:

- Are you using a Raspberry Pi? The GPU may be using HDMI I2C at the same time and interfere with your data.
- Your HDMI cable could be bad. How long is it? Is it shielded?
- You can try to decrease the I2C speed. Google for `i2c_arm_baudrate` and set it to something far lower than 400 kHz.

The EEPROM chip should be compatible considering it's command interface. There commands are the same as for P25Q40H which is known to be working.

EmpressFiona, 2022-04-05:
I had trouble the first time I tried, so I used my PC instead. The cable is brand new, too, and if it were that bad I'd expect pulling the firmware off to work less well (the difference is only a handful of bytes). I'd use the baud rate command, but that only applies to the Pi...

floppes, 2022-04-05:
If you have an oscillioscope or a logic analyzer, you can check if any other program is talking on the I2C lines.

Otherwise I recommend that you use a Raspberry Pi or the Arduino method.

EmpressFiona, 2022-04-05:
The issue turned out to be an incompatible firmware file, which was resolved once I managed to flash a known-good firmware...as far as I can tell, anyway. It might be a bit more complicated; you can consider this issue closed, but I might need to come back later if I have new findings.

I'm actually using a monitor with a related chipset, the RTD2550. Since the RTD266x series is very similar to the RTD25xx series, it seems like there's some cross compatibility. (The firmware in question was for a different RTD25xx board that someone else got to work, I didn't just use one of yours like a doofus.) Good dumps can be grabbed from RTD25xx chips through the read function, but writing seems to be an issue...

EmpressFiona, 2022-04-09:
Finally got out the Raspberry Pi to be able to test with the intended device. Same results, pretty much. I don't think the baud rate setting affects the speed on HDMI's i2c.

There is one relevant thing: I took a look inside the device at the chips on the board, and my flash chip is labeled `P25D80H`, with the `Q` replaced with a `D` - according to specifications I found, it appears to be identical but for supporting only Single and Dual IO modes, without the Quad IO mode. Could be worth a look.

However, I think my conclusion is that for whatever reason, I can get code onto this flash chip, but it will only ever run either its original firmware, or a slightly modified one. While I believe the issue *could* be on the software end, I haven't collected the necessary evidence and thus for now, it's likely out of the scope of this project. An entirely new project may be required to work with RTD25XX class chips, though it might be possible to do so with a fork of this project - but that's my responsibility, not yours.

Thank you for your time. If I see anything that might require your assistance, I'll let you know.

closed 2022-04-09 by EmpressFiona