floppes/RTD266xFlash #20: Firmware dump for Elecrow RC050 5" display

theophile, 2022-08-10:
Good evening! I'm attaching the firmware dump I pulled from this device: https://www.elecrow.com/wiki/index.php?title=RC050_5_inch_HDMI_800_x_480_Capacitive_Touch_LCD_Display_for_Raspberry_Pi/_PC/_SONY_PS4

It's a 800 x 480 display with capacitive touchscreen. Fairly popular for SBC projects.
[out.zip](https://github.com/floppes/RTD266xFlash/files/9295946/out.zip)
 
I tried to use the GUI to change the logo image. The tool warned me that it is an unknown firmware, but the autodetect appeared to be successful and it generated an out_modified.bin. However, using the python script to try to flash it back to the display, I got this error:

```
Resetting device...
Entering ISP mode...
Reading chip data...
JEDEC ID: 0x856013
ID: 0x8512
Status: 0x9c
Clearing lock bits...
Status: 0x9c
Writing sectors from file out_modified.bin which differ from sectors of file out.bin...
155648
Erasing sector at address 0x26000
Writing 256 bytes to address 0x26000
Writing 256 bytes to address 0x26100
CRC validation failed! Expected 0xb7, received 0xad
Resetting device...
```

Any ideas what I might try?

floppes, 2022-08-11:
The logo section starts at 0x261A2. Writing to 0x26000 worked, probably because there was no change in this block. The next block, starting at 0x26100, contains the changed logo data.

My guess is, that erasing did not work correctly or did not finish in time.

You can try to add `time.sleep(5)` in rtd266x\rtd266x.py in line 267 (after erasing the sector, before writing it).

It is also possible that the EEPROM ist somehow write-protected. When you read the firmware after the CRC error, is it exactly the same as before? This would be a hint that erasing had no effect.

I hope this helps!

theophile, 2022-08-11:
I'll give that a try. I had to port the script to python3 because that's the environment I have on my SBC. It's possible something in the erase function is silently failing. In further experimentation I have found that the write function will fail with this CRC error if there is any difference between what's being written and what's already there. Incidentally, if I try to use the -c option to erase the entire chip, it gives me an error saying I need to specify a file. If I specify a file, it simply prints the help dialog.

theophile, 2022-08-12:
Okay, I tried a more manual approach to trying to erase the EEPROM chip. I started a python console and manually entered the environment setup statements:
```
import sys
from rtd266x.rtd266x import RTD266x
rtd = RTD266x(3, 0x4A)
```
Then I tried issuing the chip erase command:
`rtd.erase_chip()`
This immediately produced the output "True" with no delay.

So I tried manually stepping through the function, issuing the following commands in order:
```
rtd.spi_common_command(rtd.SPI_CMD_WRITE_AFTER_EWSR, 0x01, 0, 1, 0)
rtd.spi_common_command(rtd.SPI_CMD_WRITE_AFTER_WREN, 0x01, 0, 1, 0)
rtd.spi_common_command(rtd.SPI_CMD_ERASE, 0xC7, 0, 0, 0)
```
Each one produced no output except "0," in each case with no delay and no apparent activity. Restarting the display confirms that the firmware is untouched. Not quite sure what the issue is but I'm going to try the arduino method and see if that helps.

theophile, 2022-08-12:
Arduino also cannot erase the EEPROM. Read is fine, but any write attempts produce CRC errors. I think the EEPROM chip is the P25D40H: https://datasheet.lcsc.com/lcsc/2006021533_PUYA-P25D40H-SSH-IT_C559186.pdf 

It has a hardware write protection by pulling pin 3 low. I did a quick multimeter test and that pin does appear to be pulled low. I'm going to try to desolder it and lift the leg and see if that helps.

theophile, 2022-08-12:
Bingo! I lifted the leg of pin 3 from the PCB and now I can write modified firmware. Of course, now I have to figure out where are the bitsI can modify, and what the proper resolution of the boot logo is.

floppes, 2022-08-12:
Great job!
Other people also got CRC errors when flashing their firmware. They may also have a write-protected EEPROM. Your findings will surely help them.

floppes, 2022-08-12:
I have updated the Python scripts to Python 3, Python 2 is outdated.
I've also fixed the chip erase command and added more examples in the help.

theophile, 2022-08-12:
Thanks! Btw, what is the format of the boot logo as stored in EEPROM? Is it byte array? I'm trying to figure out how to take the existing boot image and convert it back into a standard image file so I can edit it and/or determine its dimensions.

floppes, 2022-08-12:
The format for the boot logo is the same as for the fonts. It uses a run-length encoding and only two colors, represented by bits 0 and 1. The first 8 bytes contain the run-length encoding table. The next two bytes contain the length of the image data. The rest is the run-length encoded image data.
Have a look at `RTD266xFlash\RTD266xFlash\FontCoder.cs` for implementation details. You can use the class to encode and decode font/logo data.

theophile, 2022-08-12:
Thank you. Will have a look. In the meantime, I have hit a snag. To experiment, I took the factory firmware dump and changed just the logo foreground and background colors, then attempted to flash that back to the display with the updated python script with the -d option. It said:
```
Resetting device...
Entering ISP mode...
Reading chip data...
JEDEC ID: 0x856013
ID: 0x8512
Status: 0x9c
Clearing lock bits...
Status: 0x0
Writing sectors from file out_gb.bin which differ from sectors of file ../out.bin...
Erasing sector at address 0x13000
Writing 256 bytes to address 0x13000
CRC validation failed! Expected 0x8b, received 0xc8
Resetting device...
```
And now the display appears to be bricked. Even after power cycling, I can't establish an i2c connection with it. i2cdetect -y 3 produces this:
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         08 09 0a 0b 0c 0d 0e 0f 
10: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 
20: 20 21 UU 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 
30: 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f 
40: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f 
50: 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f 
60: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 
70: 70 71 72 73 74 75 76 77  
```
Arduino can't connect either. I'm thinking I'm going to have to try to solder directly to the SPI pins on the flash rom chip and figure out how to flash it directly?

theophile, 2022-08-13:
Whew. I desoldered the flash chip from the display and soldered it into a SOP-8 breakout board I had from one of those Amazon multipacks. Then I used https://github.com/nfd/spi-flash-programmer with an Arduino Pro Micro (in 3.3v mode) to directly flash the chip with the backup of the stock firmware dump. Then to remove the chip from the breakout board and solder it back onto the display. And now It's revived and working again!

I'm not sure why my earlier attempt to flash modified firmware bricked it. All I had done was change the colors of the boot logo. Am I correct that I can use any two colors I want as long as I know the 3-byte hex value for the colors? Or do the colors have to be in a pre-defined pallette too?

floppes, 2022-08-13:
Good work! I'm glad you have the skills and tools to help yourself :)

When you change the foreground and background colors of the boot logo with the GUI tool, the color palette `tPALETTE_0` is modified. It contains 16 3-byte RGB values. `tPALETTE_0` is located at offset 0x13A59 in your firmware.

The foreground color is the 5th color (at offset 12 in `tPALETTE_0`, absolute offset 0x13A65), the background color is the 15th color in the palette (at offset 42).

These changes just modify the palette, no code. Therefore I am surprised that your board did not boot anymore. Maybe the flashing was not successful and some other, relevant code bytes were corrupted.

If you are brave enough to try again, you could read the whole firmware after you flashed your modified firmware. Then compare it to what you actually flashed and check if other bytes were unintentionally changed.

theophile, 2022-08-13:
Ah, I see. I suppose there were two problems then. I didn't use the GUI tool to change the colors because I'm not positive it can recognize my firmware. Instead I loaded it into a hex editor and starting at 0x13A59, I began looking for 000000 and FFFFFF because the stock logo colors are black and white. I found them right next to each other and assumed that was the background and foreground colors and made the changes there. So that was problem #1.

Then when I went to flash it with the python script, the flash failed with a CRC error, which I feel like shouldn't have happened with write protection disabled. That's when the brick occurred. 

Maybe it was a fluke. I'm going to try to dig into the code a little further and see if I can make the tool recognize my firmware.

floppes, 2022-08-13:
I added your firmware yesterday ;)

theophile, 2022-08-13:
:flushed: Well then, let me give that a spin!

theophile, 2022-08-13:
So I used the latest build of the GUI utility. It still did not recognize the firmware. FWIW, the sha256 hash I get for my stock firmware dump is 47ff43e61156489344dad7606aebea22a670b74f375bf6e4b2c794e41c06714c.

The utility did try to autodetect the firmware and successfully flashed a modified version that changed the boot logo and background and foreground colors. However, only the foreground color actually changed, and the logo displays partially garbled. The "source" image I used was a monochrome png with a resolution of 204x72.

theophile, 2022-08-14:
I retract my previous comment about version 2.13 not recognizing my firmware. I got ahead of myself and ran the previous version by mistake. V 2.13 _does_ recognize my firmware, but so far I still can't get it to change the boot logo without it being partially corrupted/garbled. I'm thinking that maybe it's a resolution thing. I'm not sure the native resolution of the logo on my display.

I spent the day digging into FontCoder.cs to see if I could figure out how it works. I started trying to reimplement the class in python since I'm more familiar with that and don't have any experience with C#. I think though that I am in over my head.

I see that the FontCoder class has a method to decode the image from firmware and render a bmp, but it doesn't look like it's called anywhere. Is it currently possible to extract the logo from firmware and save it to a file? Maybe then I could use that as a template for a modified logo that would display correctly.

floppes, 2022-08-14:
There is a hidden "Extras" window in the GUI tool for experimentation. You can open it by double-clicking on the label "Please select the way you interface with your RTD266x chip" in the main window. Click "Decode font..." and you can display some logos I extracted from firmwares and created myself. Their binary data is in `FromFont.cs`. This uses `FontCoder.Decode()`.

I extracted your logo data and it displays like this:

[image]

The logo is internally handled as a font. Each character in a font is like a sprite with a fixed width and height. Fonts in the RTD266x firmware usually use 16x16 pixels for a character. The logos I encountered so far use 17x4 or 19x8 pixels. Your logo uses 34x8 pixels. This results in a logo size of 407x108 pixels.

When you change the logo with the GUI tool, it uses 17x4 pixels for each sprite. That's why the logo looks garbled on your display. You need to pass 34 and 8 as parameters to the constructor of `FontCoder`. You will also need to modify the validity checks when you load a graphic to accept the size 407x108. If I have more time, I will see if I can improve this generally. I will need to find a way to extract the sprite size from the firmware.

theophile, 2022-08-14:
I noticed the "extras" code but hadn't yet worked out whether it was implemented. I'll play with it. I'll see if I can modify the parameters in FontCoder.cs. I downloaded VS yesterday but hadn't spent much time with it. I've been compiling software in the command line in Linux for 25 years but never in windows.

theophile, 2022-08-15:
Okay, I've got it now! The logo resolution on this display is 408x108 and the pixel size is actually 34x6. If I change FontWidthKedei to 34 and FontHeightKedei to 6 in FontCoder.cs, the logo replacement works perfectly! Thanks for all your help on this, and thanks for this great software!

closed 2022-12-27 by floppes