floppes/RTD266xFlash #21: python script doesn't work

bernouilli, 2022-12-14:
In the file rtd266x.py, there is in line 2 :
from rtd266x.smbus import SMBus

But there is no file SMBus.py  in the directory rtd266x

floppes, 2022-12-14:
Did you run `sudo apt-get install python-smbus` as described in README.me?

bernouilli, 2022-12-14:
Yes, I installed the smbus. The error is from converting to python 3
The line
from smbus import SMBus
was converted to 
from rtd266x.smbus import SMBus

It should'nt have been converted.
I replaced this line with what it was before the converting to python3 and I managed to read the firmware.
Howeevr there is small mistake in the readme.
The commen line for reading should be:
python rtd266x_flash.py -i I2CBUS -r 524288 -f out.bin

I managed to read the firmware but wasn't able to flash a new one.
I get this message :
Resetting device...
Entering ISP mode...
Reading chip data...
JEDEC ID: 0xc22013
ID: 0xc212
Status: 0xff9c
Clearing lock bits...
Status: 0xff9e
Writing data from file freesync.bin to offset 0...
Erasing sector at address 0x0
Writing 256 bytes to address 0x0
CRC validation failed! Expected 0x9b, received 0xb2
Error: cannot program flash!
Resetting device...

I guess I'll have to try another method, maybe with windows.

floppes, 2022-12-19:
Thanks, you are right. I fixed the import statement and the readme.

Some EEPROMs are write-protected. The erase and write commands seemingly succeed and the chip does not respond with an error, but they actually do nothing. When you read the EEPROM again, it will be the same data as before.

Some EEPROMs have a pin for write-protection. You can try to lift it off and connect it with either GND or supply voltage (check the datasheet what is required to turn off the write-protection).

bernouilli, 2022-12-19:
I think that I don't have the right firmware for my board because when I try to flash the firmware I extarcted, it works fine. 
however I was wondering if it there isn't a a problem of erasing the first block and then there is the CRC error that prevent the flashing to continue. This means that the first sector is then erased and thus the board gets bricked. 
would it be possible to check the crc before erase a sector ?

floppes, 2022-12-20:
Yes, if you flash back the same firmware you read, the CRC is correct because nothing was changed in the EEPROM.

If you erase the first sector and a following write fails, your board cannot boot. However, in your case, it still seems to work. This is another proof that nothing was actually changed in the EEPROM.

Can you post a picture of your EEPROM? We can check if it has a pin connected for write-protection.

closed 2023-11-24 by floppes