floppes/RTD266xFlash #24: PCB80052 firmware unknown

dolomon, 2024-04-23:
Hi, firstly thank you so much for your dedication and hard work.

I successfully read the firmware from the above board (which has an RTD2660H) using your python tool.

When I try to open it in the editor I get "unknown firmware"

Is there any way I can work with it. I tried opening it in a hex editor but I'm not really sure what I'm looking for, I did find a sequence of bytes that seem similar when looking to disable the call to the hdmi source dialogue.
[PCB80052.zip](https://github.com/floppes/RTD266xFlash/files/15080776/PCB80052.zip)

Any help would really be appreciated, thank you in advance.

floppes, 2024-04-23:
After the error message about the unknown firmware you can select to automatically analyze the firmware. The result looks good, you can give it a try and disable the HDMI popup. If it works, I can add your firmware.

dolomon, 2024-04-24:
Hi, thank you for replying so fast, I did try that, and it may be me but it failed to detect.

dolomon, 2024-04-24:
[AT065TN14-Innolux.pdf](https://github.com/floppes/RTD266xFlash/files/15091894/AT065TN14-Innolux.pdf)

floppes, 2024-04-24:
I just published v2.14.0.0 which includes your firmware. You can give it a try.
You are right, I fixed the automatic firmware detection more than a year ago but haven't released a build with the fix. Therefore it didn't work for you. This is fixed now.

dolomon, 2024-04-25:
Thank you very much for that! I should have stated I was using the compiled binary sorry. I'm guessing if I had built from source it would have worked straight away. Thank you again, I am away but will try flashing the modified firmware at the weekend.

closed 2024-04-25 by dolomon

dolomon, 2024-04-26:
Thank you again, the tool was able to modify firmware to remove hdmi source dialogue and alter background colour. Although it could not inject a custom logo, or inject any of the included logos. The error was "too long" I'm guessing my firmware doesn't have enough space anywhere including the extra calls?

floppes, 2024-04-26:
That's good news :)
Yes, the logo in your firmware is just 56 bytes long. That's the maximum length for a new logo. Other firmware's logos are about 1500 Bytes long.
I tried to decode the logo in your firmware:
[image]
Is that your boot logo?

dolomon, 2024-04-26:
Yes it's excellent booting cleanly to a black background and not having an annoying dialogue.

I honestly didn't think there was a logo on this board, I've never seen it. Well not until you just showed us anyway.

How bizarre, the stock firmware is a blue background with yellow foreground. Unless the logo was the same colour as the background, is it xored though?

floppes, 2024-04-27:
Ah, then the logo is not enabled in your firmware. They just inserted a little dummy graphic.
The logo would have been displayed with the first two colors from your palette.

dolomon, 2024-04-27:
Ah I see, thank you for the clarification. Shame they didn't leave it in, but it's not the end of the world. Quite a tidy little board, predominantly hdmi, it does have av pins broken out but I'm guessing they stripped that from the firmware as they do not yield an image