floppes/RTD266xFlash #27: Custom logo

Mendilli2, 2024-11-03:
Hi, thanks for the this amazing tool. I am working on my own source code, everything works good except the startup logo, I am trying to change the startup logo but for weeks but can not make it

Logo is created within a header file named logo.h, I have attached the file below can you help me how to modify it, dou you have a "how to" documentation or a tool to help me change the logo

[Logo.h.txt](https://github.com/user-attachments/files/17612228/Logo.h.txt)

floppes, 2024-11-04:
I think you need to modify `tLogoTable` and adjust `LogoWidth` and `LogoHeight` in `tLogoOsdMap`. Put your encoded logo bytes into `tLogoTable` and set its width and height accordingly. You may also need to edit `tLogoPalette` if you want other colors in your logo.

To encode your custom logo, have a look at RTD266xFlash/RTD266xFlash/FontCoder.cs. You can easily modify my code to load a graphic and output the encoded bytes.

Here is an example of a custom logo I found in one of the source code repos online:

[Logo_Visatech.zip](https://github.com/user-attachments/files/17615823/Logo_Visatech.zip)

Maybe it helps you to understand what needs to be adjusted.

Mendilli2, 2024-11-04:
Thanks for the answer,

I have already worked on visatech logo on my custom source code setup,
Altough it exist in source code it is not actually working, it gives just a
blank screen (it seems to be faulty), I will try to examine fontcoder.cs, I
hope I can make it work

4 Kas 2024 Pzt 11:58 tarihinde Florian ***@***.***> şunu
yazdı:

Mendilli2, 2024-11-04:
Because I am a total noob with coding, it seems imposible for me to achieve
it by analyzing fontcoder.cs file. Would it be too much to ask you to hep
me editing the logo.h file for the logo below

şunu yazdı:

floppes, 2024-11-04:
I can see no logo file attached to your posts...

Mendilli2, 2024-11-04:
Sorry, here it is
[image]

Mendilli2, 2024-11-04:
İt will be 17x11 (204x198 px) at the center just the same as the one in logo.h file

floppes, 2024-11-04:
This is a true color graphic. You have to reduce the number of used colors to 16 or less.

Mendilli2, 2024-11-04:
is this ok?
[logo.bmp.txt](https://github.com/user-attachments/files/17624274/logo.bmp.txt)

Mendilli2, 2024-11-05:
this is 16bit color too
[logo.bmp.txt](https://github.com/user-attachments/files/17629379/logo.bmp.txt)

floppes, 2024-11-05:
I looked into this but I would need more time than I have right now. Sorry, I can't help you.

Mendilli2, 2024-11-05:
That is ok, thanks for your effort anyway, I will try to resolve it using
your fontcoder.cs file

5 Kas 2024 Sal 23:52 tarihinde Florian ***@***.***> şunu
yazdı:

closed 2024-11-09 by floppes