floppes/RTD266xFlash #28: I2C connection broken after firmware uploding

Mendilli2, 2024-11-09:
Hi, I have this source code below, after compiling it successfully with keil, I use RTD266xFlash with arduino to upload the resulting firmware to a pcb800099 board, RTD266xFlash successfully connects the board at the first place, but after I upload my new firmware, RTD266xFlash never connects the board again, İt costed me 4 driver boards so far,İt is obviouus it is something related with my source code but I can not understand the problem. Can someone please help

[RTD2662-master.zip](https://github.com/user-attachments/files/17688549/RTD2662-master.zip)

Mendilli2, 2024-11-10:
I was using the  sda, scl and ground pins of vga connector on the pcb800099 board with arduino, after uploading the firmware (on my first post), it stopped working (i2c connection can not be established any more).

 I have found this link saying that a MODIFIED HDMI CABLE can be used with RTD266xFlash tool, but there is nothing about how the hdmi cable should be modified, I want to try the hdmi connector on the pcb800099 for i2c connection, so can someone please help me about the hdmi cable option

https://githubhelp.com/floppes/RTD266xFlash

[image]

floppes, 2024-11-10:
Just have a look at the pins of an HDMI connector (https://en.wikipedia.org/wiki/HDMI): 15 is SCL, 16 is SDA and 17 is GND.

But I guess that these signals are connected to the ones from the VGA connector and it makes no difference if you use VGA or HDMI for the I²C connection.

From my experience, even after flashing garbage as a firmware, the chip is still responsible and another firmware can be flashed. I have no idea what happened to your boards...

If it really is an issue with the firmware: as far as I know, RTD266x's firmware is stored on the external EEPROM. You could use an in-circuit EEPROM flasher to flash a working firmware. If you don't have such a flasher, solder wires to the EEPROM's pins and connect them to an Arduino. There are probably Arduino sketches available to use the Arduino as an EEPROM flasher.

Mendilli2, 2024-11-10:
According to this page pcb800099 firmware has no i2c eeprom, firmware is stored in spi-flash

https://forums.adafruit.com/viewtopic.php?t=84068

floppes, 2024-11-11:
This shouldn't be a problem. SPI flash chips can be read and written easily with an Ardiono as well. If you can identify the flash chip, seach for an Ardiono sketch.

closed 2024-11-26 by Mendilli2