floppes/RTD266xFlash #30: Flasher crashes on Windows 10

Deadking2010, 2024-12-25:
I find that sometimes when trying to connect to the device or trying to upload the firmware the program just closes.  I can get to about 20% at times and then the program just closes.

Is there a preferred settings for windows?

I have also tried to run the programmer under Wine in Linux and I get a checksum error when trying to read data.

Any help would be appreciated.

floppes, 2024-12-28:
Are you using the GUI tool? If yes and you have Visual Studio installed (the community edition is sufficient), you can run it in debug mode and see if you get an exception.

There are no special settings required in Windows.

Deadking2010, 2024-12-28:
Yes I was using the GUI tool but I do not have visual studio installed.  I will install it and see what happens then.  Thanks.

Deadking2010, 2024-12-29:
I was able to install Visual Studio and run the program,  it worked flawlessly.  Only thing I can think is that I did not have .Net installed and I was prompted to by VS.  I was able to flash a new firmware for the board and can confirm the new screen works as it should.

Thanks

closed 2024-12-29 by Deadking2010